SDraw/driver_leap #6: Doesn't work at all

Silithas, 2018-08-27:
in VR, this disables roomscale tracing, so i just look around in a set point in space.

SDraw, 2018-08-28:
Hm, not enough information.
- What HMD are you using?
- What build of SteamVR (Beta or Stable) are you using?
- What drivers are loaded in SteamVR besides this one?
- Is there anything strange in '<steam_dir>/logs/vrserver.txt'?
- What Orion version are you using?

Silithas, 2018-08-31:
HMD is htvc vive
Steam VR beta update
No other drivers as far as i know of.
Honestly don't know, and too much info to read through every entry.
Orion version? What's orion?

rSh4d0w, 2018-10-31:
Orion is the software from the leap motion device which u need in order to use the device.
guess the fault here sit infront of the pc, since u don't even know how to setup the leap motion and its software. do u even own a leap motion?

seanakiyama, 2019-01-25:
It doesn't seem to work for me at all. 

I did all the steps, but alpha #8 does not seem to be working at all..
nor does the alpha 3..

the gesture checker can see my hands and registers my gestures
i can use trigger, grip, menu button
but when i enter steam vr none of that happens.

win10, steam vr 1.2.7, oculus app version 1.34.0.198261 with dk2

KiKi170987, 2019-02-26:
Hi there, I don't know if it's the right place to publish this, so sorry if it's not...

First of all, my LM works fine, I can see my hand in the visualizer and in the LM apps, the issue is with steam VR.
I have done all the steps with the latest cbuchner version and follow all the step here but I can't see the virtual controllers showing on steam VR, only my rift and the 2 cameras (and the 2 rift touch when active).
I have tried to uninstall everything and did all the steps again but same thing.

The only step I have a doubt is on: vrpathreg adddriver "path_to_leap_folder", where I have added the following folders:
[image]
Does it seems good to you ?
Btw, the first instalaltion I did placed the drivers in program file x86 but with the re-install it changed.
I really don't know where the issue come from, but I can't use my LM in VR anymore.

I'm running W10 64bits, oculus rift 1.35.0.206993 (CV1), steamVR 1.2.10.
Thank you in advance for your help.

SDraw, 2019-03-02:
@KiKi170987 You should add exactly **leap** folder to vrpathreg, because SteamVR tries to find drivers by searching in drivers' subfolders 'bin/Win32' and 'bin/Win64'.
So for you (as I can see) it's
```
vrpathreg adddriver "C:\Program Files\SteamVR Leap Motion driver\leap"
```
And don't forget about
```JSON
"activateMultipleDrivers": true,
```
in _steamvr.vrsettings_ too.

KiKi170987, 2019-03-04:
@SDraw Thank you for your answer and for your work it's great work.
I have finally solve my issue yesterday, but I don't even know how.
Before I uninstalled the drivers and steam VR the first time, these two things were ok so I didn't know where to look.
And yesterday, I decided to also uninstall steam and reinstall everything from scratch... And miracle, it worked... I don't know where the issue was at the end, but apparently, reinstalling steamVR wasn't enough.
Thank you for your support.

g-coded, 2019-05-02:
Having problems getting this working.  I confirmed pathreg(after adding it), and I get the icons in the SteamVR tray(greyed out), but can't get gestures(time out) to work.  Should this display hands/wands when working?

SDraw, 2019-05-02:
@g-coded 
- Grey icons - controller isn't connected
- Blinking green icons - hands aren't in controller view
- Green icons - all good

According to this, you need to check why it isn't connected.
- Is Orion Beta installed?
- Is LeapService running?
- Is controlled working in diagnostic visualizer or another application?

g-coded, 2019-05-02:
I have used the LM with the Blocks demo, and I start the monitor before
starting SteamVR. I had this working (hand tracking) with Driver4vr, but I
uninstalled that in case there are any conflicts.

kelemenlajos, 2019-05-03:
@g-coded I had a similar issue before (controller not starting and not showing in steamvr window). I solved this by checking the logs.
You can find yours via vrpathreg:

`g:\GAMES\Steam\steamapps\common\SteamVR\bin\win64>vrpathreg
Runtime path = G:\GAMES\Steam\steamapps\common\SteamVR
Config path = D:\Games\Steam\config
Log path = D:\Games\Steam\logs

g:\GAMES\Steam\steamapps\common\SteamVR\bin\win64>`

in the logs dir, look for:
d:\Games\Steam\logs\vrserver.txt
d:\Games\Steam\logs\vrmonitor.txt
d:\Games\Steam\logs\vrclient_leap_monitor.txt
d:\Games\Steam\logs\vrclient_Steam.txt

My solution was, i needed to use the vrpathreg adddriver command to add the driver, but after that i found some issues with the initialization of the driver, steam tried to load it twice. I double checked it was only one time in the vrpathreg. So i removed the driver path with vrpathreg removedriver and after that it started.

@SDraw But i also have a problem... My leap motion controls in vr, are showing up only a few times in application... (I don't know how and when, till yet i can't recreate it).
For now, I can see the 2 green C icons pulsating and if i raise my hand in the front of the sensor, the icons are switching to constant green (so leap motion should work), but i cant see my hands in SteamVR. I see nothing, and are not able to interact with anything. I only got to see the vive controllers in the place of my hands, if i'm switching apps. In these few secs i'm amble to move them, but didn't have time to test out any more. You can find my logs here:
https://drive.google.com/open?id=19W6w9_esHHM2WlWO5C-POfcWS9E6344o
(If you need any more logs, feel free to list them...)

g-coded, 2019-05-04:
@kelemenlajos thanks for the suggestions, I'm not sure what I'd be looking for in the log files, and the date of the log files isn't current(I ran SteamVR right before checking).  I removed the driver with cmd(icons gone), then added, and also tried SteamVR Beta/main, and no luck.  I also removed "Driver4VR" comments in the vrsettings file.  I think my next step is to uninstall SteamVR/Steam like @KiKi170987 did, but I'm not sure it's worth it.

SDraw, 2019-05-04:
@kelemenlajos Hm, in provided logs your controller is constantly being plugged and unplugged. Not actually sure, but it's worth to check cable for damage.
@g-coded Alright, I'll download Driver4VR to check compatibility with it.

kelemenlajos, 2019-05-04:
@g-coded if you share your log files, i can dig into them... Just double check the log path with vrpathreg. 
Also please do a reinstall of the leap driver like this before sending the logs:
- exit SteamVR (if it's running)
- exit Steam (if it's running)
- shutdown leap motion controller (if it's running)
- kill leap motion service via Task manager (if it's running)
- delete your leap directory from ?:\???\Steam\steamapps\common\SteamVR\drivers\
- run the streamvr_driver_leap_alpha8 installation app again.
- Repair the drivers
- download both x86,x6 drivers from https://github.com/SDraw/driver_leap/releases
- Copy them to the right directory in ?:\???\Steam\steamapps\common\SteamVR\drivers\leap\bin overwriting the files.
- Open steamvr.vrsettings in Steam/config folder and add line in "steamvr" section:
"activateMultipleDrivers": true, of it's not present
- DON'T add the driver via vrpathreg adddriver (for now)
- If you know how to restart a service, restart LeapService via Task manager or service manager. If not, simply restart your machine.
- Check the  leap motion controller, it should be static green in Notification area. If it's blinking more than 1 sec (if it's blinking it's trying to reconnect to the device), then unplug your leap motion wait 5 sec and plug it in again (don't do unplug, and instant replug, it's not good for the hardware)
- Start Steam
- Start SteamVR
- You should see 2 green C icons blinking. If you put your hands before the leapmotion about 0.5m away, the blinking should stop and the C icons should be stable green.

I played around with some VR apps last night and it's worth to get it going. Before you take the reinstall path, try the leap motion built in visualizer (you can access it via the leap motion control panel in the Notification area. Right click and start the visualiser). This also should be sent to the VR headset, so you can check the latency and check if there are any disturbing IR sources that are blocking leap motion from working as it should. This also can be a sign for signal loss, please use a sorter or better quality or shielded usb cable.

@SDraw as for my issue:
My setup definitely works. But i'm only able to use this controller via, leap motion visualiser (which is independent of steamVR):
https://photos.app.goo.gl/oy1Zg3ckYyQWzksJA (turn down the volume, my phone amplified the fan nose to an insane level)

Or in the SteamVR void space:
https://photos.app.goo.gl/aMCE4Fv7DpHJrr277 (turn down the volume, my phone amplified the fan nose to an insane level)
Or in SteamVR setting screen:
https://photos.app.goo.gl/G4uR8cMoD371iAhk7 (turn down the volume, my phone amplified the fan nose to an insane level)

It wont show up on any application. I already set the controller configuration, to vive_controller on all apps i tried. Any ideas what can be wrong on my side?

kelemenlajos, 2019-05-04:
> @kelemenlajos Hm, in provided logs your controller is constantly being plugged and unplugged. Not actually sure, but it's worth to check cable for damage.

I already sorted this out, but thanks for pointing it out ;)

g-coded, 2019-05-04:
@SDraw - I have tried both disabling/uninstalling the Driver4VR app with no luck.  Leap as Vive controllers do work using Driver4VR(but gestures are not supported)
@kelemenlajos - I followed the steps, downloaded the x86 drivers(was only using 64 before).  I did notice the SteamVR drivers folder didn't have the 'leap' folder which I didn't delete before, so I copied those.  I also copied 'leap.vrsettings'(after failing without it).  I'm still getting gray icons in SteamVR.

The leap motion is working fine, natively(Blocks), Driver4VR, visualizer.  I'm not sure if I should add the path at this point?  Also note, I have rebooted my machine after all this just to be sure a process wasn't preventing something from working.

kelemenlajos, 2019-05-04:
@g-coded could you please send your logs?

g-coded, 2019-05-04:
[Here's](https://pastebin.com/7rvkzzkp) the leap monitor to start with, looking at it, it may have something to do with my hmd(Pimax), there is also a "Unable to register named resource path" listing both the C drive and Steam directory.

kelemenlajos, 2019-05-04:
@g-coded hmm... that's not nice...

first, you have more settings than needed i think:
```
Sat May 04 2019 07:17:48.400 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\SteamVR\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.400 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\SteamVR\drivers\00vrinputemulator\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.400 - [Settings] Load Default Json Settings from C:\Program Files\Pimax\SteamVRSupport\drivers\aapvr\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.400 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\MixedRealityVRDriver\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.401 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\SteamVR\drivers\htc\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.401 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\SteamVR\drivers\lighthouse\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.401 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\SteamVR\drivers\null\resources\settings\default.vrsettings
```

These are sospicious:
```
Sat May 04 2019 07:17:48.400 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\SteamVR\drivers\00vrinputemulator\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.400 - [Settings] Load Default Json Settings from C:\Program Files\Pimax\SteamVRSupport\drivers\aapvr\resources\settings\default.vrsettings
Sat May 04 2019 07:17:48.400 - [Settings] Load Default Json Settings from S:\Steam\steamapps\common\MixedRealityVRDriver\resources\settings\default.vrsettings
```
- as far as i understand you have a pimax, so the row with pimax should be okay
- mixedreality is from windows vr setting, that also should be okay
- but i would doublecheck maybe they won't go together. But 00vrinputemulator i never heard of... and based on this: https://github.com/matzman666/OpenVR-InputEmulator it's quite possible could interfere with leap motion...

the second issue is with this line
```
Sat May 04 2019 07:17:48.402 - [Settings] Load Json Settings from C:\Users\rdwyer.DESKTOP-C63EHO2\AppData\Local\openvr\steamvr.vrsettings
```
This should be your steamvr.settings file from your steam base dir. But it's sourced from somewhere else... Try to add 
"activateMultipleDrivers": true
to this file (to the steamvr section)

3rd, this is also sounding fishy for me... Why it's trying to get the data from SteamVR Leap Motion driver? It should use the second path... (without mentioning it i guess...)
```
Sat May 04 2019 07:17:48.441 - Unable to register named resource path leap with 'C:\Program Files\SteamVR Leap Motion driver\leap\resources' because it is already set to 'S:\Steam\steamapps\common\SteamVR\drivers\leap\resources'
```

Last, this should be somehow connect to pimax, i'm almost all cases find this error code in pimax related sites:
```
Invalid input type click for controller (/user/head/proximity)
```

g-coded, 2019-05-04:
I deleted OVRIE, but that didn't help, I removed the 'null' driver folder.  I removed/added the leap driver using vrpathreg(to try to remove it from looking in 2 locations).  I've had "activateMultipleDrivers": true enabled for as long as I've used SteamVR.  I don't think there could be issues with Pimax/WMR drivers as there is no other issues with my setup.  I'm still getting the same gray icons?

kelemenlajos, 2019-05-04:
And what changed in your logs?

" I've had "activateMultipleDrivers": true enabled for as long as I've used SteamVR."
okay, but in the log you sent me sad, that you read the steamvr.vrsettings file from here: C:\Users\rdwyer.DESKTOP-C63EHO2\AppData\Local\openvr\steamvr.vrsettings
did you check the file?

kelemenlajos, 2019-05-05:
@SDraw i think i found my issue...
Seems like the leap drivers hand controller is interfiring with the oculus controller in steamVR
if the oculus controller is active, and i'm putting it down to use leap input, i got many errors like this:
"Sun May 05 2019 12:40:31.333 - Invalid input type position for controller (/user/hand/right/input/trackpad)"
is there a way to somehow manually disable oculus controllers, if i using leap controls only?

kelemenlajos, 2019-05-05:
also, is there a way to remap what hand gesture is doing what on the virtual vive controller?

g-coded, 2019-05-05:
@kelemenlajos - I did check the file in that location.  Pimax recently released a new driver that has a 'Vive exclusive' option, which I'll see if that solves any compatibility errors with this driver.

Also for the remap question, I was looking for any way to get this working and looked into the 'leap.vrsettings' file, which you should be able to swap the 'button' with the gesture:

"leap_gestures" : {
      "GrabPressed" : "GestureWithin:LowerFist[0.5,1.0]",
      "MenuPressed" : "GestureWithin:FlatHandPalmTowards[0.5,1.0]",
      "SystemPressed" : "GestureWithin:FlatHandPalmAway[0.5,1.0]",
      "TouchpadAxis[0]" : "GestureAxis:ThumbPositionX[-1.0,-1.0]",
      "TouchpadAxis[1]" : "GestureAxis:ThumbPositionY[-1.0,-1.0]",
      "TouchpadPressed" : "GestureWithin:Thumbpress[0.5,1.0]",
      "TouchpadTouched" : "GestureWithin:Thumbpress[0.0,0.5]",
      "TriggerAxis[0]" : "GestureAxis:TriggerFinger[0.0,1.0]",
      "TriggerAxis[1]" : 0,
      "TriggerPressed" : "GestureWithin:TriggerFinger[0.5,1.0]"
   },

kelemenlajos, 2019-05-05:
Woooow! You're the best @g-coded, thanks

> @kelemenlajos - I did check the file in that location. Pimax recently released a new driver that has a 'Vive exclusive' option, which I'll see if that solves any compatibility errors with this [...]

kelemenlajos, 2019-05-05:
@g-coded leap.vrsettings file should go here right?
Steam\steamapps\common\SteamVR\drivers\leap\bin\Win64\

SDraw, 2019-05-05:
Sorry for late response. I'll check what can cause "invalid input" log message.
About remapping buttons, JSON block "leap_gestures" does nothing at all. It should be a feature in parent project, but hasn't been implemented and just floating around. I've moved away from this feature to game profiles that are hardcoded in source code to adapt for specific games.

g-coded, 2019-05-05:
@kelemenlajos - I think? ....I didn't realize that I had the files in 2 locations until you pointed it out(don't know if that's typical).  That is where I copied the files over to though.

g-coded, 2019-05-05:
@SDraw - I tried the 'Vive only games option' in the new Pitool driver, this seems to have fixed the issue!  It didn't appear to at first, and I am able to toggle the option off and it still works.  I am getting 2 sets of 'leap controller' icons(one grey/one green), which I can live with.  The 'controllers' won't render until I turn the Vive wands on/off though, is this normal?  Also, I can only get 'menu/trigger' gestures to work, I don't know if this is a SteamVR profile thing though.

gamerofallgames, 2019-06-28:
Hey there Sdraw. When ever i try and run steamvr with this installed. The index controllers don't show up it is just the htc wands. I have tryed everything on here. It would be nice if u could help

SDraw, 2019-06-28:
@gamerofallgames My bad, I've forgot to create 'Win32' and 'Win64' folders inside 'bin' in AppVeyor auto-build script. Release will be updated in 15 minutes.

gamerofallgames, 2019-06-28:
@SDraw tried the reinstall with the new win 32 and win 64 folders. still didn't work. the controllers were still vive wands. I even changed the control_config from vive to index. same result. steam vr still displays the vive wands and has gestures connected to the vive wands. I even added it using the vrpathreg and made sure that i had multipledrivers= true.

SDraw, 2019-06-29:
@gamerofallgames Could you send 'Steam/logs/vrserver.txt' here?

gamerofallgames, 2019-06-29:
@SDraw  sure no problem, also if you need to know my setup it is riftcat plus a leap motion.

[vrserver.txt](https://github.com/SDraw/driver_leap/files/3342131/vrserver.txt)

SDraw, 2019-06-29:
@gamerofallgames As I can see, driver is loaded from 'C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\leap'. Is control_config.xml from this folder edited for sure?

gamerofallgames, 2019-06-29:
@SDraw  OMG i feel so dumb. i just checked and i accidentally capitalized the I in index and that created a error LOL. thanks SDraw!

gamerofallgames, 2019-06-29:
@SDraw sorry to be a bother again. but i just went through the tutorial to test the controllers and the key mapping, but when they remove your controllers to show the virtual hand models, or even when they give the virtual hand models to you in the first place. the hands don't show up. IDK if it is because the models aren't in the correct file or whatever.

SDraw, 2019-06-30:
@gamerofallgames It's strange behaviour of SteamVR. If controllers are disconnected or calibrating (in case of this driver - hands out of Leap Motion controller frustum), some input pathes are invalidated, especially skeleton pathes. Easy solution: hold your hands in front of Leap Motion controller when application is launching.

TripRodriguez, 2019-07-03:
> Also please do a reinstall of the leap driver like this before sending the logs:
>
> * exit SteamVR (if it's running)
> * exit Steam (if it's running)
> * shutdown leap motion controller (if it's running) [...]

Thank you for those instructions.  I followed them and now it's working!  Only problem is SteamVR is now showing four green c hexes, two work.  Maybe because of having put in  the x64 and x86 both?

Anyway, thanks!

RedBrumbler, 2019-07-06:
I can't seem to be able to get a pointer in steam home, but my hands are tracked when I activate the steam menu in there, and it seems it's detecting gestures (still no pointer though)

what am I doing wrong here?

dpaget21, 2019-07-26:
> @g-coded if you share your log files, i can dig into them... Just double check the log path with vrpathreg.
> Also please do a reinstall of the leap driver like this before sending the logs:
>
> * exit [...]

I followed all these steps and I finally got the Blinking C's to show up, but the leap motion doesn't reconize my hands.  Only once and awhile will it go solid green and show a controller but I have never been able to move my hands and have it track.

XenCorus, 2019-08-27:
I have a rather weird problem. I'm trying to pair it with Trinus VR, and it acts really weird. In loading screen and void, it shows that it's tracking my hands, but in SteamVR itself it just doesn't show anything. I've also tried VRChat, and instead of showing up my hands moving or the controllers, there's just 2 flying balls that are supposed to be my hands. I did everything by the instructions, but it seems that something is not right. I really hope someone's gonna help me out here, because it's ticking me off already.
[vrserver.txt](https://github.com/SDraw/driver_leap/files/3543666/vrserver.txt)
[vrmonitor.txt](https://github.com/SDraw/driver_leap/files/3543667/vrmonitor.txt)
[vrclient_leap_monitor.txt](https://github.com/SDraw/driver_leap/files/3543668/vrclient_leap_monitor.txt)
[vrclient_VRChat.txt](https://github.com/SDraw/driver_leap/files/3543669/vrclient_VRChat.txt)

SDraw, 2019-08-27:
@XenCorus Sorry, but I can't help you with pairing it with Trinus VR, because it rejects to work on my setup.

XenCorus, 2019-08-27:
@SDraw Still, thanks for the reply. I guess I will try to fix it on my own, maybe some random action will do the trick (and not make everything worse in the process), or maybe someone else will answer here.

closed 2020-09-11 by SDraw