SDraw/driver_leap #10: Can you map a hand gesture with a keyboard

lazzonn76, 2019-07-08:
can you map a hand gesture with a keyboard button of your choice?  Thank you very much

SDraw, 2019-07-13:
The problem is that in this case key presses can be detected globally. I'll check watchdog interface in openvr to see if it's possible to detect key presses only if VR application is in focus.

lazzonn76, 2019-07-13:
Thank you so much everything works perfectly I'm testing it with Psvr on the pc and I perfectly emulate the valve index controllers like you do, I noticed an excessive sensitivity in recognizing gestures like the touchpad but otherwise everything works well done

SDraw, 2019-10-02:
Asking for clarification. Do you want to be able to override gestures data from CGestureMatcher or override gestures of a specific game?

lazzonn76, 2019-10-02:
> Asking for clarification. Do you want to be able to override gestures data from CGestureMatcher or override gestures in a specific game
It would be very useful to be overwriting CagestureMatcher so that you can use a gesture with a custom button thanks

SDraw, 2019-12-16:
Sorry, I can’t think of a way how to implement this.

closed 2019-12-16 by SDraw