SDraw/driver_leap #9: Thumbstick Direction with Index Mode

kool601, 2019-07-08:
Maybe have it like the touchpad circle gesture, except instead of using your index you use your thumb?

lazzonn76, 2019-07-08:
Now it works fine for me, I managed to understand the files to copy and the gestures work, I just have to change the controllers I want to put those of the index in place of others, your leap can connect it to a USB 3.0 port direct without a hub to improve performance

SDraw, 2019-07-14:
I don't think that it will be good solution because thumb is directed to opposite hand palm very often.

kool601, 2019-07-14:
Ah, yeah that makes sense. I hope someone can come up with a good gesture for thumbstick direction. ^w^

SDraw, 2019-09-14:
I will add new configuration line in control_config.xml to enable thumbstick direction from keyboard with arrow and numpad keys.

SDraw, 2019-09-15:
Implemented in d48b3c97, release will be updated after auto-build.
If NumLock is active, use arrow keys for left hand controller, Num2/8/4/6 for right hand controller.

closed 2019-09-15 by SDraw

SDraw, 2019-09-15:
Implemented in e52397d1.
Thumbstick direction is set for left hand controller with arrow keys and numpad keys (Num2/8/4/6) for right controller as long as NumLock is active.

lazzonn76, 2019-09-15:
Thank tou very much Sdraw🙏