SDraw/driver_leap #11: Anyways to check what gesture got triggered?

203Null, 2019-07-13:
Gesture is really glitchy for me. From leap motion monitor it tracks fine. But at relaxed hand position, grip and touch pad are randomly pressed down. So is there's a way for me to test it? 

Also in VRchar, fingers are not individually tracked for me.

Hardware: HTC Vive
Onion versionb 4.0.0
No other steamVR driver installed
SteamVR version is stable release.

Thank you for your hard works in this. I really appreciate it.

203Null, 2019-07-13:
Also I used index mode

203Null, 2019-07-14:
I just tried 0.0.8 and it works, so maybe profile failed to switch?

SDraw, 2019-07-14:
First of all, driver was never intended to be a full replacement of real controllers, but just a possiblity of emulation. Leap Motion controller tries to predict transformation of fingers that aren't visible, leading to strange presses. Specificaly about touchpad, there is something in Leap SDK 4.0.0 that has changed transformation of thumbs, making angle bending calculation return not accurate result. This behaviour is in my todo list.
For VRChat I recommend you to stick with Vive emulation for now, because gesture detection activation doesn't work in game with left controller's thumbstick press as it should be for unknown reasons.
Driver version 0.0.8 doesn't have Valve Index controller emulation. ¯\_(ツ)_/¯

203Null, 2019-07-14:
In vive mode gestures doesn't works too

203Null, 2019-07-18:
Only launch from steam client works, using the shourcut (steam://rungameid/438100) somehow doesn't work

closed 2019-07-18 by 203Null