SDraw/driver_leap #14: Gestures too sensitive, request parameter

AfromD, 2019-08-21:


SDraw, 2019-10-02:
Slowing down gesture recognition seems to be not possible because it's strictly connected with fingers data (direction, bending, angles beetween different fingers). Using configurable parameter for this can lead to gestures never reach full or zero recognition value.  
Dead zone sounds like a good solution, but current structure of CGestureMatcher doesn't have any access to configuration data.

OFWModz, 2020-01-20:
I'd like to see something like this too.

PickRelated, 2020-05-15:
@SDraw any way to tweak ON/OFF gesture latch? For example:
1. Thumbpress > 0.9 - Latch ON
2. Thumbpress < 0.2 - Latch OFF
It seems to me that currently we get something like `> 0.5 = ON`, `< 0.5 = OFF`. This causes a lot of triggering noise.

I hope you get my point.

closed 2020-12-19 by SDraw