SDraw/driver_leap #16: Hand rotation is off and tracking is very scattered

AIREYVR, 2019-09-09:
i seem to be having a problem where my hands are rotated inward too much on my coulus and the tracking is super jittery and looks nothing like it does in the veiwer

AIREYVR, 2019-09-09:
Tracking was an issue on my end but hand rotations on index emulation isnt proper

SDraw, 2019-09-14:
Hm, could you provide some screenshots on this issue?
From mathematical point of view position and rotation of controllers are calculated right, so it's possible that rendermodels definitions in JSON files (valve_controller_knu_1_0_left/right.json) for controllers are made with offsets.

asktoby, 2019-11-24:
I'm having this issue on Index also.
I've done a video to demonstrate it:
https://www.youtube.com/watch?v=oG_kHEl1vAo&feature=youtu.be

asktoby, 2019-12-11:
I believe this should no longer be marked "waiting response" but I do not have permission to remove the label

SDraw, 2019-12-16:
I still think that it's models offset causing this missalignment.

Maxuos, 2020-01-02:
SDraw, first of all thank you for all of your efforts... great stuff man!
Been testing on my Leap as well in SteamVR+VRChat, mounted on HMD. 
Everything is smooth.. the only issue I am facing is my hands are not aligned properly.
In leap visualizer hands are aligned properly, in VRChat my hands angles are off, any ideas how to sort it out?

allybam, 2020-03-17:
Same issue as Maxuos, my hands are also off angle. What is the correct X Y Z position?

havok2-htwo, 2020-05-22:
Hello,
is the project still in progress? It looks to me as if one line of code has been forgotten just before the finish line. Unfortunately, I have not yet received the project in compliance. I'm not a full-time programmer, but shouldn't it be enough to set a rotation offset to X,Y,Z before the handover? Everything works fine, but the hands are just 45° outwards and up I think. Also the hands are about 20cm apart, even if I press my real index fingers together. In the Config there are already offset values, but they don't hurt. Couldn't you solve this in the same way with a Config? 

Left Hand - Offset Rotation x,y,z
Left Hand - Offset Position x,y,z
Right Hand - Offset Rotation x,y,z
Right Hand - Offset Position x,y,z

Thanks!

SDraw, 2020-06-13:
Refer to #38.

closed 2020-06-13 by SDraw

asktoby, 2020-06-13:
I just tested again using the latest release driver_leap_1.0-174 and the issue is still there.
Am I supposed to manually enter the offsets in the config?