SDraw/driver_leap #17: Index not working

TheCatRegime, 2019-10-01:
Hello!, I just started using this driver and it is amazing! But it only works with the HTC and not with valve index. I'm using the Samsung Odyssey hmd  headset. Do you know what the problem may be? Thanks a lot.

SDraw, 2019-10-01:
Could you provide 'vrserver.txt' file from '<Steam_installation_folder>/logs' folder?

TheCatRegime, 2019-10-02:
[vrserver.txt](https://github.com/SDraw/driver_leap/files/3682551/vrserver.txt)

spayne, 2019-10-02:
I  was curious so I had a look at your vrserver.txt.  

What I see are a couple things that should be cleaned up.  This may or may not resolve your issue but might simplify solving it:
1. SteamVR thinks the leap driver is registered in two locations and SteamVR is refusing to load the second driver.  I don't think this is a good idea because you may be installing updates into the wrong folder, so I think you should fix this.  I added some details below on how to do that.
2. Windows MR installation is corrupt.

For 1.  The reason I believe that is because of this line: 

> Wed Oct 02 2019 13:47:50.094 - Skipping duplicate external driver leap in "D:\leap" because a driver of that name was already found at "C:\Program Files (x86)\Steam\steamapps\common\SteamVR\drivers\leap"

For 2 the reason is this line:

> Wed Oct 02 2019 13:47:50.123 - C:\Program Files (x86)\Steam\config\vrappconfig\steam.app.719950.vrappconfig - Errors when parsing manifest file:

Note that 799950 is the app id for windows mr for steam. https://steamdb.info/app/719950/

To fix 2: the minor issue, the easiest fix may be to uninstall and reinstall windows mr for steam.

To fix 1, @SDraw has a section about vrpathreg at https://github.com/SDraw/driver_leap#troubleshooting

So in your case, you need to run vrpathreg from the command line to remove the extra driver.  e.g. if the one in D:\leap is the wrong one then I'd open up a cmd shell and:
1. run "C:\Program Files (x86)\Steam\steamapps\common\SteamVR\bin\win32\vrpathreg.exe"
this will show you the ones you have installed.
2. then run "C:\Program Files (x86)\Steam\steamapps\common\SteamVR\bin\win32\vrpathreg.exe" removedriver   [path to the one you want to remove].  E.g. if the D:\leap is the wrong one, then:
"C:\Program Files (x86)\Steam\steamapps\common\SteamVR\bin\win32\vrpathreg.exe" removedriver   "D:\leap" should work.
3. run step 1 again to ensure only one driver is listed there.  

Beware that vrpathreg.exe is case sensitive.

closed 2019-12-16 by SDraw