SDraw/driver_leap #20: Should fingers be actuating in SteamVR home?

asktoby, 2019-10-18:
I am able to use gestures (close index finger for trigger, make fist for grip) but my finger model isn't moving like it does when I use an Index Controller.
Is it supposed to?
Video here:
https://youtu.be/iVDYfzMwmic

asktoby, 2019-10-18:
Worked it out.

Edited:
C:\Program Files\SteamVR Leap Motion driver\leap\cfg

    <param name="emulated_controller" value="index"/> 
    <param name="skeleton" value="true"/> 
    <param name="trackingLevel" value="full"/> 

Thanks!

SDraw, 2019-11-07:
Also hold your hands in front of Leap Motion controller during SteamVR/SteamVR Home start. Sometimes applications don't use skeletal data interfaces if controllers are out of reach (for us, hands are out of detection area).

closed 2019-11-07 by SDraw