SDraw/driver_leap #21: Leap Motion + SteamVR -> OpenVR Shutdown immediately after application run

FlorianMathis, 2019-11-06:
Hi all,

I'm facing an error and could not figure out the source of the problem. The combination of both (the leap motion and a htc vive controller (or vive tracker)) causes an automatically "OpenVR Shutdown". Anyone faced this problem and has a solution/workaround?

Best regards,
flo

SDraw, 2019-11-07:
I've updated driver to latest OpenVR SDK, there are few changes that are connected with applications.
Try to use auto-builded release 157.
If nothing helps, post 'vrserver.txt' file from <Steam>/logs folder here.

FlorianMathis, 2019-11-12:
Thank you @SDraw  - will take a look soon!

FlorianMathis, 2019-11-19:
@SDraw don't know what exactly this error caused but it works now, thanks!

closed 2019-11-19 by FlorianMathis