SDraw/driver_leap #22: Tracked hands are twitchy

koray509, 2019-11-17:
i have tried recalibrating my leap motion controller , i am emulating inndex controllers but same happens with vive controller emulation , in diagnoustics visualizer there is no twitching , when in a game the gestures track fine but the hands are twitchy and does not match the locations of my real hands , i am using the beta version of steam vr and the twitching is less when only tracking one hand.
thanks for reading

havok2-htwo, 2020-05-30:
Here I fixed it. You need Visual Studio 2013 that is not compatible with Windows 10. I have done this in a virtual machine with Windows 7. I was working on a version for DCS World 2.5.6 for my own and I thought its useful for you too: https://workupload.com/file/wZmDQPRq2YA

In the zip file is the edited source code with my comments and also the compiled 32 & 64 bit version. Have fun.

Here us a Video using it with DCS World 2.5.6: https://youtu.be/54bvY_K34yE

SDraw, 2020-06-13:
Refer to #38 .

closed 2020-06-13 by SDraw