SDraw/driver_leap #24: Update Leap Motion SDK to V4.0.0 Build 52238

DJPlaya, 2019-12-11:
There is a new Version of the Leap SDK which allows the usage of multiply Leap Motion Controllers, it would be pretty nice if you could update to that Version.
Described here > http://blog.leapmotion.com/multiple-devices/
Download here > https://github.com/leapmotion/UnityModules/raw/feat-multi-device/Multidevice%20Service/LeapDeveloperKit_4.0.0%2B52238_win.zip

SDraw, 2019-12-16:
Currently driver processes data only from one controller, and more effort is needed to implement data mixing from several controllers.
Could you check if this updated version behaves correctly with driver?

DJPlaya, 2019-12-17:
You mean the general Leap Driver? It does indeed, but beside the calibration Tool, there isent a Software that Supports multiply Controllers in that Release.
I have tried this Project in the Past, the Source may help you > https://github.com/paluka/Multi-Leap

Tutorial on how to build a Hand from Tracking Data > https://forums.leapmotion.com/t/building-a-new-hand-with-your-own-frame-data-a-tutorial/8034
Relevant Information on how to implement multiply Leaps > https://forums.leapmotion.com/t/multiple-devices-on-getting-frames-from-one-device/7578
Implementation in Unreal > https://github.com/leapmotion/LeapUnreal/releases/tag/v3.4.0
How to use Multiply Devices in Unreal > https://github.com/leapmotion/LeapUnreal/tree/feat-multi-device

SDraw, 2020-06-13:
Usage of build 52238 doesn't change tracking for driver, even though driver still uses only one controller device to get data from.

closed 2020-06-13 by SDraw

DJPlaya, 2020-06-13:
There is probably an Misunderstanding. The Driver build 52238 only enables the usage of multiply Controllers, the tracking Data has to be mixed manually.