SDraw/driver_leap #27: Can't seem to install for some reason, troubleshooting instructions do nothing

InsaneGrox, 2019-12-22:
So I've been trying to get this driver installed, but it doesn't get detected by steamVR, however I did read the troubleshooting stuff, and for some reason, vrpathreg commands are just asking how I want to open a file, and don't do anything else, I can't open command prompt in folders so I'm forced to use powershell.
[image]

SDraw, 2019-12-22:
You don't need to run vrpathreg if you're extracting to 'drivers' folder of SteamVR, only additional line in 'steamvr.vrsettings' is required.

InsaneGrox, 2019-12-22:
I tried that, but the driver doesn't load despite being in the drivers folder, I have orion drivers running and my device turned on, but no hands or controllers appear.

SDraw, 2019-12-22:
Hm, could you send vrserver.txt from 'Steam/logs'?

InsaneGrox, 2020-01-08:
oh my fucking god, my steamVR was in safe mode...

closed 2020-02-10 by SDraw