SDraw/driver_leap #29: Thumbstick control (Index)

SWigl, 2019-12-28:
Hello, I have an important discussion regarding Thumbstick control.  In most games, Thumbstick is more important than Trackpad.  And using the keyboard to move is not very convenient.  Can I change the way I control Thumbstick to the way I control Trackpad?  Using your finger is more convenient than the keyboard.  Or maybe you can add a quick switch?

Julian-MJK, 2020-01-23:
A better way to handle thumbstick & trackpad control with the Index would be to perform a gesture to switch between controlling the thumbstick and the trackpad, with your hand. 

For example: Form a C with one hand and press it against the flat palm of your other hand _(like the "T" system-button gesture, but forming a C with the other hand.)_

**Summary: A gesture to switch betweeen thumbstick & trackpad controls**

This could totally remove the need for a keyboard, making leap-motion index emulation as an alternative very usable. _After all, the index controller feature of this program is likely what a majority of people will want, over the normal vive-wands emulation._

crazyzack53, 2020-01-31:
Has there been any work on this suggestion, because it would really help in game

kool601, 2020-02-10:
I hope this gets implemented, it sounds much better than the current thumbstick implementation.

SDraw, 2020-06-13:
Sorry, but I can't do this suggestion.

closed 2020-06-13 by SDraw