SDraw/driver_leap #32: Hand alignment is majorly off

Zuma206, 2020-03-11:
Hi SDraw, I recently downloaded your driver, and the finger tracking itself worked fine. However, the VR hands where my real wrists are, and when I turn my hands, they move even further away from where they are meant to be.

Additional Info:
HMD: Oculus Rift S
SteamVR Version: 1.10.26
driver_leap Version:  1.0-170

crazyzack53, 2020-05-07:
Has this problem been solved yet, because I have the same problem

Zuma206, 2020-05-11:
> 
> 
> Has this problem been solved yet, because I have the same problem

Unfortunatly, I could not solve this issue.

Julian-MJK, 2020-05-11:
Perhaps this could be solved with a calibration sequence, similarly to what OpenVR Space Calibrator does (or perhaps we could just use that in itself? It's on GitHub, I don't know about it's licensing or usability. Here's a video demonstrating it: https://youtu.be/JnX8NK-lsPY?t=280)

If this were fixed, well it would frankly be fantastic

Zuma206, 2020-05-12:
> Perhaps this could be solved with a calibration sequence, similarly to what OpenVR Space Calibrator does (or perhaps we could just use that in itself? It's on GitHub, I don't know about it's [...]

I sold my leap motion now so I wont be able to try this, but I think it should work!

havok2-htwo, 2020-05-30:
Here I fixed it. You need Visual Studio 2013 that is not compatible with Windows 10. I have done this in a virtual machine with Windows 7. I was working on a version for DCS World 2.5.6 for my own and I thought its useful for you too: https://workupload.com/file/wZmDQPRq2YA

In the zip file is the edited source code with my comments and also the compiled 32 & 64 bit version. Have fun.

Here us a Video using it with DCS World 2.5.6: https://youtu.be/54bvY_K34yE

SDraw, 2020-06-13:
Possibility to set own offset is now added in b1a949aa.

closed 2020-06-13 by SDraw