SDraw/driver_leap #37: Null driver setting not working

chenj209, 2020-05-19:
I tried to use this driver_leap with null driver setting since I don't have a VR headset right now. However, SteamVR does not recognize the controller for some reason.

Here is the steps to reproduce:
On windows computer, I have leap motion orion 4.0.0+52173 version installed
1. add "activateMultipleDrivers" : true, "forcedDriver": "null", to near the top of the "steamvr" section of your steamvr.vrsettings config file.
2. set driver null to be enabled in \SteamVR\drivers\null\resources\settings\default.vrsettings
3. Create 'leap' folder in '<SteamVR_folder>/drivers'
4. Grab latest release archive
5. Extract archive to '<SteamVR_folder>/drivers/leap'
6, Add line in section 'steamvr' of '<Steam_folder>/config/steamvr.vrsettings' file: "activateMultipleDrivers": true,
7. Open console as administrator in '<SteamVR_folder>/bin/win32' (or '<SteamVR_folder>/bin/win64') and execute command:  vrpathreg adddriver "path_to_leap_folder"

The result is no controller is recognized

chenj209, 2020-05-20:
[image]

SDraw, 2020-06-13:
Seems like you've mixed first and second methods of installation. With first method you don't need to add path to driver with **vrpathreg**. Try to remove added path in **vrpathreg** and check if SteamVR isn't running in safe mode.

closed 2020-09-11 by SDraw