SDraw/driver_leap #38: I solved the Hand-Offset

havok2-htwo, 2020-06-02:
The hands have a huge offset in rotation. Here I fixed it. I was working on a version for DCS World 2.5.6 for my own and I thought its useful for you too: https://workupload.com/file/wZmDQPRq2YA

In the zip file is the edited source code with my comments and also the compiled 32 & 64 bit version. Have fun.

Too see it works correct - here us a Video using it with DCS World 2.5.6: https://www.youtube.com/watch?v=q6CwQciWyHg

Julian-MJK, 2020-06-04:
Oh god this is incredible! This is literally something I've wanted for my Leap Motion since I bought it, and I was so bummed to see that the tracking was off for me, so I'm definitely pulling the thing back up after this.

@SDraw Could you have a look at this? We should create a pull request to merge the changes, a bunch of Leap Motion users are probably a little bit eager to start finger tracking again after all the finger tracking news on the Oculus Quest.

crazyzack53, 2020-06-05:
Wait, how do I install this, where am I supposed to put the files?

havok2-htwo, 2020-06-05:
@crazyzack53 
https://github.com/SDraw/driver_leap

here you can find a manual how to install the original programm. In the path where it is installed replace the files with mine. Its pretty simple im the end :)

CyberThund3r, 2020-06-06:
@havok2-htwo Awesome thank you so much for this! it's fixed for everything i tried but for some reason VRChat still has the hand offset issue i guess because it uses it's own profile or something, what can i do to change it?

SDraw, 2020-06-06:
@havok2-htwo Looks good, but to keep everything in order I'll make possibility to use your changes through editing of configuration file (better offsets settings for short). After some testing I'll inform you here.

havok2-htwo, 2020-06-06:
> @havok2-htwo Looks good, but to keep everything in order I'll make possibility to use your changes through editing of configuration file (better offsets settings for short). After some testing I'll inform you here.

Hey @SDraw 
thank you!  Yes, I'm a noob about github and I haven't had the time to watch it with the config file.  My hope was that you do exactly that and at least don't have to do the code search.  What I haven't figured out yet is how exactly the axes are arranged. Because its not x,y,z for the rotation. I've compiled about 50 times until the values (nearly) fit.  Maybe you can find a better variant for the User Config which is more understandable.

officialnochill, 2020-06-07:
Is this the bug where your hands teleport to the ground when out of Leap motion view? 

Would love to see that fixed

havok2-htwo, 2020-06-07:
Hello @officialnochill 
no this only fixes the rotation of the hands.  These currently do not correspond to the real rotation of the real hands in the virtual world.  I am not sure if this could be fixed.  @SDraw couldn't you tell this thing to output the last position when the hand becomes "invalid"?

SDraw, 2020-06-13:
@havok2-htwo Merged in b1a949aa. Based on your changes, your new settings should be
```XML
  <setting name="leftHandOffset" value="-0.025 0.08 -0.08"/>
  <setting name="leftHandOffsetRotation" value="0.181 -0.067 -0.118 0.974"/>
  <setting name="rightHandOffset" value="0.025 0.08 -0.08"/>
  <setting name="rightHandOffsetRotation" value="0.181 0.067 0.118 0.974"/>
```
Tested for myself, seems same as in your video.

havok2-htwo, 2020-06-13:
Maybe you can leave a comment line on how to convert xyz values to that 4 quarternation numbers? Well done!

crazyzack53, 2020-06-13:
I'm still confused, where exactly am I supposed to input the values?

havok2-htwo, 2020-06-13:
> I'm still confused, where exactly am I supposed to input the values?

> I'm still confused, where exactly am I supposed to input the values?

I did not testet it yet, but there is a config file where you can change some values. I thing he added the new 4 values to that config file. Its in the installation path. Was an .xml?

CyberThund3r, 2020-06-18:
@SDraw i compiled the latest version and i've tried changing the new settings but doesn't seem to be doing much even though i have restarted SteamVR each time i did (idk how to get it changing dynamically 🤔), i want to fix the hand facing forward orientation as you can see in this [video ](https://streamable.com/rv0tg5) it's not where the fingers should be like in the index controller which also causes VRC avatar arm IK snapping when rotating it in some angles..

https://streamable.com/rv0tg5

SDraw, 2020-09-11:
@CyberThund3r Should be fixed since 2630833.

closed 2020-09-11 by SDraw