SDraw/driver_leap #39: Question: Does the index contrller emulation support finger splay?

RileyGuy, 2020-06-07:
As the title says, I'm getting a Leap Motion in several days and I wanted to know if I can perform finger splaying with the index controller emulation.

SDraw, 2020-06-07:
As non-English speaker I can't quite understand meaning of "finger splaying". But if you're talking about calculating angle between neighbor fingers, then yes, it is. Here is an example of sample Unity project with SteamVR skeletal input and this driver in Index emulation mode with desktop orientation.
https://mega.nz/#!cZZjUCqQ!Dax7-i0hbbXSobW-MtDK0JwxhipxbzTJiVf3BTlvH40

RileyGuy, 2020-06-08:
Splaying is moving your fingers closer or further from eachother, yes. The ability to increase the distance between them and angle your fingers from side to side. The video answers my question, thank you very much! I'll be using your driver once I get my leap. One further question though, can I use my vive wands *and* use this driver at the same time? As in, being able to put my wands down and switch to the leap and then back again.

SDraw, 2020-06-13:
Yes, you can. Currently you can press Ctrl+P and Ctrl+O with active NumLock to enable/disable left and right hand controller. The only thing is what controllers are used for input up to game itself. If game was coded right for switching between active controllers, then everything will be fine.

closed 2020-09-11 by SDraw