SDraw/driver_leap #46: Automatic tracking mode

FaxMac, 2020-07-27:
Basically, my issue is in this reddit posthttps://www.reddit.com/r/leapmotion/comments/hspznd/how_do_i_manually_change_the_tracking_mode_from/ but to keep it short, does this driver automatically change tracking mode to vr? and if not , would it be possible to do it without manually going to the VR visualiser and changing it manually?

SDraw, 2020-07-28:
By default [tracking mode is set to HMD](../blob/master/driver_leap/CServerDriver.cpp#L150). You can [change its orientation in `settings.xml`](../blob/master/resources/settings.xml#L6) before SteamVR launch. Currently there is no way to change orientation while SteamVR is running.

SDraw, 2020-09-11:
In 930d291 I've implemented orientation change on configuration reload. Perhaps, I'll add keyboard shortcut to switch it faster than with manual file editing.

closed 2020-12-20 by SDraw