SDraw/driver_leap #49: Doesn't work with SteamVR?

Lythinca, 2020-08-15:
Hi! I followed the instructions in the readme, but wasn't able to get it working with SteamVR. I installed the "VR" version of Leap Motion Orion (as the link in the readme brings me to a page that asks if I want to use the Desktop or VR version of leap orion).

I'm not using beta version of SteamVR, SteamVR is not in safe mode. I've edited the SteamVR.vrsettings file as well.

I notice that whenever I launch SteamVR, it overwrites the steamvr.vrsettings file and removes the line that I added in for multiple drivers.

I can see that my leap motion works in VR by choosing "Visualizer" in the Leap Motion application, but steamVR itself doesn't see it as a controller.

ShootingKing-AM, 2020-08-16:
Copy paste your `SteamVr settings` here, you probably have an extra "," somewhere, and if possible try to copy paste `Steam/logs/vrserver.txt` file

Lythinca, 2020-08-17:
Worked! Thank you!

closed 2020-09-11 by SDraw