SDraw/driver_leap #50: HandReset set true doesn't work

Kokka0426, 2020-08-30:
the handreset worked previously but i had to reinstall windows and thus all my apps again, and for some reason this time handreset isnt working. i need it working again as i use an xbox controller for movement and menu navigation but while the index controllers are 'in range' the menu will appear on my avatars hand instead of stationary in front of me. any help is much appreciated!

Kokka0426, 2020-08-31:
just a message so you know i'm still active here.

Kokka0426, 2020-09-05:
still active

Kokka0426, 2020-09-07:
boop

SDraw, 2020-09-09:
Got it

Kokka0426, 2020-09-09:
ᐠ(  ᐢ ᵕ ᐢ )ᐟ yay!

do you need anything from me? files or logs?

SDraw, 2020-09-09:
Nah.
I've made `handsReset` to be applied no matter what controllers are emulated and what game is launched in 694885844c1bf175094158654597b217033f84ef. Try 1.0-181 from releases and set `handsReset` to `true`. I hope that you don't use Vive controllers emulation in VRChat.

Kokka0426, 2020-09-09:
ありがとう！ Thank You! ᕕ( ᐛ )ᕗ

Kokka0426, 2020-09-10:
just got home to be able to test it out, and sadly same issue persists with new bin and resource folders exported into leap folder under divers of steamvr

SDraw, 2020-09-11:
Strange.
Anyway, I've made a little change in 930d291. Try 1.0-182, don't forget to set `handsReset` to `true`.

Kokka0426, 2020-09-11:
thanks so much for the prompt reply! i'll hive it a shot as soon as i get home from work and let you know if it works now. if not feel free to ask me for any files, logs, settings info etc. i'd hate for you to do extra work if it turns out its just me being dumb.

Kokka0426, 2020-09-11:
still a no go. do i need to replace more than the contents of the steamvr/drivers/leap folder?

SDraw, 2020-09-12:
Have you edited `resources/settings.xml` and changed value for `handsReset` to `true`? I've tested it on my side and menu in game appears in front of me 🤔

SDraw, 2020-09-12:
Seems like that it's on VRChat side. I can confirm that year ago if controller was marked as out-of-range, game reset corresponding avatar hand. Now it just remembers last controller transformation but doesn't reset avatar hand. Strange decision of VRChat dev team, even though game knows that controller is out-of-range and enables one-handed game controls.

closed 2020-09-28 by SDraw