SDraw/driver_leap #51: Oculus Touch emulation?

rccDave, 2020-09-04:
HTC Vive and Valve are listed.  I tried it just the same with an Oculus Rift 2 and Touch Controllers.  In the SteamVR home room I did not see my hands via LeapMotion Controller, but I did, of course, see the Touch Controllers.

I saw somewhere that someone had to turn off their controllers to get your driver and the LeapMotion Controller to be recognized.  I tried that.  Still no hands (and no controllers that time).

I hope you have plans to add Oculus Touch emulation.  I want to use it in X-Plane to flip switches, etc.

SDraw, 2020-09-09:
Oculus Touch emulation can be added, for sure. But it will only work inside SteamVR and games launched from it, but not Oculus VR environment.
And driver was designed in mind that you can turn off your controllers to make game switch to emulated controllers. But in case of real Oculus Touch controllers, it is special case that can't be resolved and your only way is to remove batteries from them to switch to emulated ones.

rccDave, 2020-09-09:
I understand.

X-Plane uses SteamVR when it runs, even when launched from Oculus, so I think the emulator would work.

If controller menus and all are there, I don't mind taking the batteries out of the Touch controllers.  I do that anyway when I'm not using to extend battery life.  (The controllers seem to have some power consumption even when "off". Probably for touch monitoring.)

I'm happy to help try it out and give feedback if you add Oculus Touch to it.  How can I help?

P.S. Oculus put hand tracking in the Quest, but not Rift 2.  It's not as good as LeapMotion's sensor, but it's something and may indicate the next update to Rifts could have it.  That's all speculation.  It could be a while yet, so having your emulator would be handy now.

closed 2020-12-12 by SDraw

rccDave, 2020-12-12:
SDraw,
I see "done" and "closed".  Does this mean that LeapMotion now works with a Rift S through Steam VR?

SDraw, 2020-12-12:
Part of milestone. I will make commit with it soon.

rccDave, 2020-12-13:
Got it.  Thanks.  Can't wait to try it with X-Plane through Steam VR.

rccDave, 2021-01-08:
@SDraw 
Happy new year!  Checking in to see if  this is ready to try with Rift S on Steam VR and X-Plane.  I'm looking for "Commit" in the status.  Correct?