SDraw/driver_leap #56: Tracking issues with Pimax + HMD mounted leap

PylotLight, 2020-12-16:
Hey,

2 issues I've noticed and not sure if they have easy solutions.

1. The hands move with the HMD, not on their own in 3d space, so if I move my head, the hands follow then readjust some seconds later.
2. This brings me to the next issue, there's a huge lag in the tracking, from when I move my hands, to when it's registered in SteamVR home is about 2-3 seconds and it just doesn't feel accurate at all.

Is there any way to improve this?

Regards,
Lochy.

SDraw, 2020-12-19:
Are you using AMD cpu?

PylotLight, 2020-12-19:
Nope!
i7 8700k
EVGA 2080Ti XC Gaming
32GB 3222mhz

chrisnovello, 2020-12-20:
AMD CPU over here, though I also tested and found the release before current (driver_leap_1.0-189) does work, and doesn't have the reported latency or parented-to-head issue, so just using that for the moment 

(tho I suspect I will eventually test bringing a hand rig in via some means.. maybe overlay / openxr / or reality tab, tho havent had time to look into it or think it through for my need yet!)

PylotLight, 2020-12-20:
Much less lag on the previous version for me too!
(Although the teleport commands don't seem to work in 'the lab' for some reason.)

SDraw, 2020-12-20:
>     1. The hands move with the HMD, not on their own in 3d space
The thing is, this is what it's supposed to be. 

>     2. This brings me to the next issue, there's a huge lag in the tracking
This issue affects first one. Try to check controller state in Leap Motion Diagnostic Visualizer, perhaps your data rate is low. Moreover, this issue was present when I was making CLeapPoller and made simple error in code. Strange, but it shouldn't be present now.

> AMD CPU over here, though I also tested and found the release before current (driver_leap_1.0-189) does work, and doesn't have the reported latency or parented-to-head issue, so just using that for the moment 

There is strange issue that compiled driver on Intel cpu works with problems on AMD cpu systems. Appveyor uses Intel-based cloud compilation, so there is no possible way for me to complie driver on AMD cpu.

PylotLight, 2020-12-21:
There's other apps I've used where it doesn't work that way, so not sure if it's possible to update the location quicker to keep them in place, as you move your head, then it puts them back where they were.
[image]

lefufu, 2020-12-29:
hello. Same here: 2 to 3 seconds of lag...

sampan2000, 2021-01-12:
Hey,
I think the 2 issues are caused by the lag.
I have an AMD 5800x CPU,and have the lag too.I tried to complie driver on my computer,and copied the build files to the driver folder,the lag still remains.
The driver_leap_1.0-189 works fine for me.
Regards,
Sam

sampan2000, 2021-01-18:
> Hey,
> I think the 2 issues are caused by the lag.
> I have an AMD 5800x CPU,and have the lag too.I tried to complie driver on my computer,and copied the build files to the driver folder,the lag still remains.
> The driver_leap_1.0-189 works fine for me.
> Regards,
> Sam
I opened the solution with VS 2019 and upgraded the build toolset to vs2019(v142),the issues resolved.

marcotrani, 2021-01-25:
It seems same here, LM Visualizer is quickly responsive but in Steam VR (DCS game based on Steam VR too) the lag is very high.
Cpu I9 9900K ... Someone has solved that problem? 

Thanks

sampan2000, 2021-01-25:
> It seems same here, LM Visualizer is quickly responsive but in Steam VR (DCS game based on Steam VR too) the lag is very high.
> Cpu I9 9900K ... Someone has solved that problem?
> 
> Thanks

complie using VS2019

marcotrani, 2021-01-25:
> complie using VS2019

Thansk Sampan 2000 for fast and cool reply... Unfortunately I'm very low skilled about VS2019 and I really don't know how to do what you said :-/ ...

If there is a guide to do that I will be happy to learn it, right now for me is غير مفهوم :-)

sampan2000, 2021-01-25:
> > complie using VS2019
> 
> Thansk Sampan 2000 for fast and cool reply... Unfortunately I'm very low skilled about VS2019 and I really don't know how to do what you said :-/ ...
> 
> If there is a guide to do that I will be happy to learn it, right now for me is غير مفهوم :-)

Follow the steps in the [README.md](https://github.com/SDraw/driver_leap/blob/master/README.md)，but with VS2019 and chose bulid tool V142 when asked to upgrade the solution.

marcotrani, 2021-01-25:
> Follow the steps in the [README.md](https://github.com/SDraw/driver_leap/blob/master/README.md)，but with VS2019 and chose bulid tool V142 when asked to upgrade the solution.

Thanks again, with a friend's help we finally have it work with no lag also on our systems!

Also no lag in fly playing DCS... And we noticed could be more useful a bind to disable temporarily the hand gesture... Anyone knows if it's possible or will be added?

Really thanks a lot for develop and support!

PylotLight, 2021-01-29:
Rebuilt the solution in vs2019 for my 8700k, but lag is still present, unlike the previous release, just fixing that would be dope.
Also the lab tutorial doesn't seem to recognise any gestures, not sure if I should log that as a separate issue or not @SDraw