SDraw/driver_leap #57: Steam VR can´t see the controls

Lynxai, 2021-01-05:
Hi, I followed the steps but i must miss something:

Pimax 5K+ with Stereo IR 170 working fine in visualizer and Pimax Experience software, https://www2.leapmotion.com/v4.1-sir170-windows-sdk installed, "leap" folder in steamapps\common\steamvr\drivers but steamvr (BETA) can´t see the controls.

Lynxai, 2021-01-05:
`{"collisionBounds":{"CollisionBoundsColorGammaB":189,"CollisionBoundsFadeDistance":0.60000002384186,"CollisionBoundsColorGammaG":0,"CollisionBoundsGroundPerimeterOn":true,"CollisionBoundsColorGammaR":255,"CollisionBoundsColorGammaA":237},"audio":{"onRecordDevice":"{0.0.1.00000000}.{50637989-1dd3-421a-876b-98f026f7b1db}","enablePlaybackDeviceOverride_Pimax VR, Inc._Pimax 5K Plus":true,"playbackDeviceOverride_Pimax VR, Inc._Pimax 5K Plus":"","playbackDeviceOverrideName_Pimax VR, Inc._Pimax 5K Plus":"","onPlaybackDevice":"{0.0.0.00000000}.{79082bf8-5066-4a25-ae00-13963356b8ad}","enableRecordingDeviceOverride_Pimax VR, Inc._Pimax 5K Plus":false,"activePlaybackMirrorDevice":"{0.0.0.00000000}.{a9366513-4679-480e-85bd-a8d744e18c77}","enablePlaybackMirror":true},"steam.app.611670":{"resolutionScale":416,"appName":"The Elder Scrolls V: Skyrim VR"},"GpuSpeed":{"gpuSpeedVersion":2,"gpuSpeed2":389,"gpuSpeed8":333,"gpuSpeedVendor":"NVIDIA GeForce GTX 1060 6GB","gpuSpeed7":387,"gpuSpeed1":393,"gpuSpeedRenderTargetScale":0.1799999922514,"gpuSpeed6":399,"gpuSpeedHorsepower":391,"gpuSpeed5":391,"gpuSpeedDriver":"26.21.14.3602","gpuSpeed0":396,"gpuSpeed9":392,"gpuSpeedCount":10,"gpuSpeed3":387,"gpuSpeed4":398},"steamvr":{"installID":"5034583263377773123","supersampleScale":0.24,"activateMultipleDrivers": true,"mirrorViewDisplayMode":2,"showPerfGraph":true,"lastVersionNotice":"1.13.10","allowAsyncReprojection":false,"showMirrorView":false,"maxRecommendedResolution":8192,"supersampleManualOverride":true,"showAdvancedSettings":true,"overlayRenderQuality_2":2,"mirrorViewEye":2,"background":"#FF000000","allowInterleavedReprojection":false,"lastVersionNoticeDate":"1594678162"},"DesktopUI":{"pairing":"569,271,800,600,0","settings_desktop":"406,271,800,600,0"},"openvr.tool.steamvr_environments":{"motionSmoothingOverride":0},"dashboard":{"position_2":0,"lastAccessedExternalOverlayKey":"matzman666.AdvancedSettings"},"LastKnown":{"HMDManufacturer":"Pimax VR, Inc.","HMDModel":"Pimax 5K Plus"}}`

Lynxai, 2021-01-05:
[https://file.io/CCAAEjcDLdTs](url)

SDraw, 2021-01-05:
Have you installed x86 or x64 release?

Lynxai, 2021-01-05:
x64. I have just tried x86 version too, no luck.

closed 2021-01-09 by Lynxai