SDraw/driver_leap #59: Lag issue with Reverb G2 HMD Mounted in DCS game

marcotrani, 2021-01-25:
Hi! I've installed your driver SDRaw (also added Orion original software / CDBuchner Driver_Leap / Visual C++ 2015 MS Redistribuitable) and all works fine in Orion 3d software (like cube or painter) , it starts fine with Steam VR (latest beta release and stable release both) , very efficently and fast.

When I use it in DCS it comes very slow, when I move my hand the in-game hands moves but with about 500msec delay... 

If in DCS I use hand controller with original Reverb G2 controller the hands go very fast and no delay from real movement.

I tried also to emulate "vive" "oculus" and "index" too but the problem is still the same.

Can you help me to avoid this delay?
Thanks

Adil3tr, 2021-01-27:
I'm dealing with the same issue but I can't figure out what's causing it.

PylotLight, 2021-01-29:
There's already an issue posted for this, you really didn't need to submit a duplicate issue.